d-m-bailey/cvc #224: Build instructions clarification

mguthaus, 2021-02-25:
I think the builld instructions should require running autoconf -i as well before configure?

d-m-bailey, 2021-02-27:
@mguthaus Thanks for the suggestion. 
In my experience, running autoconf on the installation machine occasionally resulted in an error when re-yacc-ing the parser. 
So far, I have not had any problems installing on linux machines running the configure script as is, but let me know if you run into problems and I'll look into it.

mguthaus, 2021-02-27:
Hi Mitch,

There is no configure script committed, just the configure.ac file.

Matt

d-m-bailey, 2021-02-28:
@mguthaus Thanks for the clarification.

The github page has instructions for installing from the **distribution release** which includes the configure script, but no instructions for installing from the git repo.

I should add a file to repo to explain that more clearly.