d-m-bailey/cvc #233: Running CVC on the extracted GDS spice

Manarabdelaty, 2021-09-06:
Hi @d-m-bailey 

I am trying to run CVC on the spice extracted from the gds. CVC first complained on missing models: 
```
ERROR: No model match sky130_fd_sc_hd__a21oi_1/X4 X sky130_fd_pr__pfet_01v8_hvt ad=0p pd=0u as=0p ps=0u w=1e+06u l=150000u
ERROR: No model match sky130_fd_sc_hd__a21oi_1/X5 X sky130_fd_pr__nfet_01v8 ad=0p pd=0u as=0p ps=0u w=650000u l=150000u
Missing models

sky130_fd_pr__nfet_01v8 X
sky130_fd_pr__pfet_01v8_hvt X
```
So, I gave it the primitive devices cdl files but I got the following parsing error: 
```
 /openLANE_flow/designs/spm/runs/spm/results/cvc/spm.cdl:13379.1-7: syntax error, unexpected SUBCKT
Fatal error:Could not parse /openLANE_flow/designs/spm/runs/spm/results/cvc/spm.cdl
```
I am attaching the cdl file along with the cvc logs. 
[cvc.tar.gz](https://github.com/d-m-bailey/cvc/files/7115151/cvc.tar.gz)

d-m-bailey, 2021-09-06:
@Manarabdelaty The extracted spice file needs to be modified to be more spice-like to run CVC. X-devices need to be converted to mosfets, etc. Are you an efabless employee? If so, I can show you where I keep the scripts.

mithro, 2021-09-06:
@d-m-bailey I would highly suggest making these scripts publically available under an open source license so anyone can use CVC with the OpenLane flow output?

Manarabdelaty, 2021-09-06:
@d-m-bailey Yes, I am an efabless employee. It would be very helpful if you can show me the scripts and if you don't mind we can get them integrated into openlane like @mithro suggested.

d-m-bailey, 2021-09-06:
@mithro Right, the current integration on CVC into openlane is somewhat limited. From the CVC standpoint, the ideal input would be created by converting the verilog gate level netlists to spice and then adding the spice libraries. I haven't figured out how to do that yet, so the current implementation uses the maglef extracted layout with the spice library cells. However, I believe that it did not handle hierarchical verilog for some reason. 
What I working on now is using the device level extracted netlist, but since the device names and some net names do not match the netlist, debugging the results can be non-trivial.

mithro, 2021-09-06:
Release early, release often :-)

The great thing about open source is that other people can play with your half finished stuff and help you finish it.