d-m-bailey/cvc #236: error: implicit instantiation of undefined template 'std::array<unsigned int, 5>'

yurivict, 2021-12-28:
```
In file included from cdlParser.cc:41:
In file included from cdlParser.yy:32:
In file included from ./Cvc.hh:62:
/usr/include/c++/v1/vector:428:64: error: implicit instantiation of undefined template 'std::array<unsigned int, 5>'
        __alloc_traits::destroy(__alloc(), _VSTD::__to_address(--__soon_to_be_end));
                                                               ^
```

Did you forget to include ```<array>```?

clang-13

d-m-bailey, 2021-12-28:
Thanks for the heads up! 

Just to be clear, are you following the installation instructions or trying to compile using another method?
```
download tarball from release page https://github.com/d-m-bailey/cvc/releases.
extract
cd cvc-<version>
autoreconf -vif
./configure --disable-nls [--prefix=<install_directory>]
make install
```

yurivict, 2021-12-28:
The FreeBSD port that I develop essentially follows these steps under the hood.