d-m-bailey/cvc #238: The install target fails: TypeError: an integer is required (got type bytes)

yurivict, 2021-12-28:
```
13386 INFO: Building PYZ (ZlibArchive) /disk-samsung/freebsd-ports/cad/cvc/work/cvc-1.1.0-4-gd172016/src_py/build/check_cvc/PYZ-00.pyz
Traceback (most recent call last):
  File "/usr/local/bin/pyinstaller", line 33, in <module>
    sys.exit(load_entry_point('PyInstaller==3.5', 'console_scripts', 'pyinstaller')())
  File "/usr/local/lib/python3.8/site-packages/PyInstaller/__main__.py", line 111, in run
    run_build(pyi_config, spec_file, **vars(args))
  File "/usr/local/lib/python3.8/site-packages/PyInstaller/__main__.py", line 63, in run_build
    PyInstaller.building.build_main.main(pyi_config, spec_file, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/PyInstaller/building/build_main.py", line 844, in main
    build(specfile, kw.get('distpath'), kw.get('workpath'), kw.get('clean_build'))
  File "/usr/local/lib/python3.8/site-packages/PyInstaller/building/build_main.py", line 791, in build
    exec(code, spec_namespace)
  File "check_cvc.spec", line 21, in <module>
    pyz = PYZ(a.pure, a.zipped_data,
  File "/usr/local/lib/python3.8/site-packages/PyInstaller/building/api.py", line 98, in __init__
    self.__postinit__()
  File "/usr/local/lib/python3.8/site-packages/PyInstaller/building/datastruct.py", line 158, in __postinit__
    self.assemble()
  File "/usr/local/lib/python3.8/site-packages/PyInstaller/building/api.py", line 128, in assemble
    self.code_dict = {
  File "/usr/local/lib/python3.8/site-packages/PyInstaller/building/api.py", line 129, in <dictcomp>
    key: strip_paths_in_code(code)
  File "/usr/local/lib/python3.8/site-packages/PyInstaller/building/utils.py", line 652, in strip_paths_in_code
    consts = tuple(
  File "/usr/local/lib/python3.8/site-packages/PyInstaller/building/utils.py", line 653, in <genexpr>
    strip_paths_in_code(const_co, new_filename)
  File "/usr/local/lib/python3.8/site-packages/PyInstaller/building/utils.py", line 660, in strip_paths_in_code
    return code_func(co.co_argcount, co.co_kwonlyargcount, co.co_nlocals, co.co_stacksize,
TypeError: an integer is required (got type bytes)
gmake[3]: *** [Makefile:452: check_cvc] Error 1
gmake[3]: Leaving directory '/disk-samsung/freebsd-ports/cad/cvc/work/cvc-1.1.0-4-gd172016/src_py'
gmake[2]: *** [Makefile:402: all-recursive] Error 1
gmake[2]: Leaving directory '/disk-samsung/freebsd-ports/cad/cvc/work/cvc-1.1.0-4-gd172016'
gmake[1]: *** [Makefile:343: all] Error 2
gmake[1]: Leaving directory '/disk-samsung/freebsd-ports/cad/cvc/work/cvc-1.1.0-4-gd172016'
===> Compilation failed unexpectedly.
```

yurivict, 2021-12-28:
This failure is the same in 1.1.0 and in the current rev. d172016

d-m-bailey, 2021-12-28:
I've only been able to get the python/kivy GUI for error analysis (check_cvc) to compile on a very few versions of Linux. Most of the time I have to copy the results to a PC and run under IDLE.

The README file addresses this.

I may make the GUI compile separate since it usually doesn't work. Thanks for the report.

d-m-bailey, 2022-04-30:
@yurivict I'm removed the gui installation from the default. You should not get this error now.

closed 2022-04-30 by d-m-bailey