d-m-bailey/cvc #239: clang-13 prints many warnings/errors

yurivict, 2021-12-29:
These warnings likely indicate real errors:
```
./CCvcExceptions.hh:49:101: warning: address of stack memory associated with local variable 'myMessage' returned [-Wreturn-stack-address]
        const char* what() const noexcept { string myMessage = "Invalid terminal: " + errorMessage; return myMessage.c_str(); }
                                                                                                           ^~~~~~~~~
./CCvcExceptions.hh:57:96: warning: address of stack memory associated with local variable 'myMessage' returned [-Wreturn-stack-address]
        const char* what() const noexcept { string myMessage = "Model error: " + errorMessage; return myMessage.c_str(); }
                                                                                                      ^~~~~~~~~
./CCvcExceptions.hh:70:98: warning: address of stack memory associated with local variable 'myMessage' returned [-Wreturn-stack-address]
        const char* what() const noexcept { string myMessage = "Database error:" + errorMessage; return myMessage.c_str(); }
                                                                                                        ^~~~~~~~~
./CCvcExceptions.hh:78:95: warning: address of stack memory associated with local variable 'myMessage' returned [-Wreturn-stack-address]
        const char* what() const noexcept { string myMessage = "Fatal error:" + errorMessage; return myMessage.c_str(); }
                                                                                                     ^~~~~~~~~
./CCvcExceptions.hh:91:95: warning: address of stack memory associated with local variable 'myMessage' returned [-Wreturn-stack-address]
        const char* what() const noexcept { string myMessage = "Queue error:" + errorMessage; return myMessage.c_str(); }
                                                                                                     ^~~~~~~~~
./CCvcExceptions.hh:99:95: warning: address of stack memory associated with local variable 'myMessage' returned [-Wreturn-stack-address]
        const char* what() const noexcept { string myMessage = "Power error:" + errorMessage; return myMessage.c_str(); }
                                                                                                     ^~~~~~~~~
./CCvcExceptions.hh:107:100: warning: address of stack memory associated with local variable 'myMessage' returned [-Wreturn-stack-address]
        const char* what() const noexcept { string myMessage = "Resistance error:" + errorMessage; return myMessage.c_str(); }
                                                                                                          ^~~~~~~~~
7 warnings generated.
```

There are also these errors:
```
In file included from CVirtualNet.cc:24:
In file included from ./CVirtualNet.hh:30:
./mmappable_vector.h:151:10: error: no member named '_M_impl' in 'std::vector<CVirtualNet, mmap_allocator_namespace::mmap_allocator<CVirtualNet>>'
                        Base::_M_impl._M_finish = Base::_M_impl._M_start + n;
                              ^
./mmappable_vector.h:92:4: note: in instantiation of member function 'mmap_allocator_namespace::mmappable_vector<CVirtualNet>::_M_set_finish' requested here
                        _M_set_finish(n);
                        ^
CVirtualNet.cc:82:2: note: in instantiation of member function 'mmap_allocator_namespace::mmappable_vector<CVirtualNet>::mmap_file' requested here
        mmap_file(theSourceVector.size());
        ^
In file included from CVirtualNet.cc:24:
In file included from ./CVirtualNet.hh:30:
./mmappable_vector.h:139:30: error: no member named '_M_get_Tp_allocator' in 'std::vector<CVirtualNet, mmap_allocator_namespace::mmap_allocator<CVirtualNet>>'
                        A *the_allocator = &Base::_M_get_Tp_allocator();
                                                  ^
CVirtualNet.cc:84:2: note: in instantiation of member function 'mmap_allocator_namespace::mmappable_vector<CVirtualNet>::remmap_file_for_read' requested here
        remmap_file_for_read();
        ^
```

d-m-bailey, 2022-04-30:
@yurivict Finally got around to fixing some of these errors. If you have time, could you try with the current repo and let me know the results?

yurivict, 2022-04-30:
The ```<array>``` header is missing:
```
In file included from cdlParser.cc:41:
In file included from cdlParser.yy:32:
In file included from ./Cvc.hh:62:
/usr/include/c++/v1/vector:428:64: error: implicit instantiation of undefined template 'std::array<unsigned int, 5>'
        __alloc_traits::destroy(__alloc(), _VSTD::__to_address(--__soon_to_be_end));
                                                               ^
/usr/include/c++/v1/vector:371:29: note: in instantiation of member function 'std::__vector_base<std::array<unsigned int, 5>, std::allocator<std::array<unsigned int, 5>>>::__destruct_at_end' requested here
    void clear() _NOEXCEPT {__destruct_at_end(__begin_);}
                            ^
/usr/include/c++/v1/vector:465:9: note: in instantiation of member function 'std::__vector_base<std::array<unsigned int, 5>, std::allocator<std::array<unsigned int, 5>>>::clear' requested here
        clear();
        ^
/usr/include/c++/v1/vector:497:5: note: in instantiation of member function 'std::__vector_base<std::array<unsigned int, 5>, std::allocator<std::array<unsigned int, 5>>>::~__vector_base' requested here
    vector() _NOEXCEPT_(is_nothrow_default_constructible<allocator_type>::value)
    ^
./CCircuit.hh:35:7: note: in instantiation of member function 'std::vector<std::array<unsigned int, 5>>::vector' requested here
class CCircuit {
      ^
/usr/include/c++/v1/__tuple:219:64: note: template is declared here
template <class _Tp, size_t _Size> struct _LIBCPP_TEMPLATE_VIS array;
                                                               ^
In file included from cdlParser.cc:41:
In file included from cdlParser.yy:32:
In file included from ./Cvc.hh:39:
In file included from ./CvcTypes.hh:28:
In file included from /usr/include/c++/v1/string:511:
In file included from /usr/include/c++/v1/string_view:179:
In file included from /usr/include/c++/v1/__string:57:
In file included from /usr/include/c++/v1/algorithm:653:
/usr/include/c++/v1/memory:874:58: error: implicit instantiation of undefined template 'std::array<unsigned int, 5>'
            _VSTD::__libcpp_deallocate((void*)__p, __n * sizeof(_Tp), _LIBCPP_ALIGNOF(_Tp));
                                                         ^
/usr/include/c++/v1/__memory/allocator_traits.h:280:13: note: in instantiation of member function 'std::allocator<std::array<unsigned int, 5>>::deallocate' requested here
        __a.deallocate(__p, __n);
            ^
/usr/include/c++/v1/vector:466:25: note: in instantiation of member function 'std::allocator_traits<std::allocator<std::array<unsigned int, 5>>>::deallocate' requested here
        __alloc_traits::deallocate(__alloc(), __begin_, capacity());
                        ^
/usr/include/c++/v1/vector:497:5: note: in instantiation of member function 'std::__vector_base<std::array<unsigned int, 5>, std::allocator<std::array<unsigned int, 5>>>::~__vector_base' requested here
    vector() _NOEXCEPT_(is_nothrow_default_constructible<allocator_type>::value)
    ^
./CCircuit.hh:35:7: note: in instantiation of member function 'std::vector<std::array<unsigned int, 5>>::vector' requested here
class CCircuit {
      ^
/usr/include/c++/v1/__tuple:219:64: note: template is declared here
template <class _Tp, size_t _Size> struct _LIBCPP_TEMPLATE_VIS array;
                                                               ^
In file included from cdlParser.cc:41:
In file included from cdlParser.yy:32:
In file included from ./Cvc.hh:62:
/usr/include/c++/v1/vector:374:52: error: implicit instantiation of undefined template 'std::array<unsigned int, 5>'
        {return static_cast<size_type>(__end_cap() - __begin_);}
                                                   ^
/usr/include/c++/v1/vector:466:57: note: in instantiation of member function 'std::__vector_base<std::array<unsigned int, 5>, std::allocator<std::array<unsigned int, 5>>>::capacity' requested here
        __alloc_traits::deallocate(__alloc(), __begin_, capacity());
                                                        ^
/usr/include/c++/v1/vector:497:5: note: in instantiation of member function 'std::__vector_base<std::array<unsigned int, 5>, std::allocator<std::array<unsigned int, 5>>>::~__vector_base' requested here
    vector() _NOEXCEPT_(is_nothrow_default_constructible<allocator_type>::value)
    ^
./CCircuit.hh:35:7: note: in instantiation of member function 'std::vector<std::array<unsigned int, 5>>::vector' requested here
class CCircuit {
      ^
/usr/include/c++/v1/__tuple:219:64: note: template is declared here
template <class _Tp, size_t _Size> struct _LIBCPP_TEMPLATE_VIS array;
                                                               ^
In file included from cdlParser.cc:41:
In file included from cdlParser.yy:32:
In file included from ./Cvc.hh:62:
[... 164 lines cut ...]
        __alloc_traits::deallocate(__alloc(), __begin_, capacity());
                        ^
/usr/include/c++/v1/vector:551:5: note: in instantiation of member function 'std::__vector_base<std::array<unsigned int, 5>, std::allocator<std::array<unsigned int, 5>>>::~__vector_base' requested here
    ~vector()
    ^
./CCircuit.hh:35:7: note: in instantiation of member function 'std::vector<std::array<unsigned int, 5>>::~vector' requested here
class CCircuit {
      ^
/usr/include/c++/v1/__tuple:219:64: note: template is declared here
template <class _Tp, size_t _Size> struct _LIBCPP_TEMPLATE_VIS array;
                                                               ^
[... cut ...]
CCircuit.cc:257:45: note: in instantiation of member function 'std::vector<std::array<unsigned int, 5>>::operator[]' requested here
                                if ( (*circuit_ppit)->devicePrintCount_v[device_it][theErrorSubIndex] > 0 ) {
                                                                        ^
/usr/include/c++/v1/__tuple:219:64: note: template is declared here
template <class _Tp, size_t _Size> struct _LIBCPP_TEMPLATE_VIS array;
                                                               ^
14 errors generated.
```

d-m-bailey, 2022-04-30:
@yurivict Does adding the the following line fix the problem?
```
diff --git a/src/Cvc.hh b/src/Cvc.hh
index 8df45c1..c01d04b 100755
--- a/src/Cvc.hh
+++ b/src/Cvc.hh
@@ -59,6 +59,7 @@ extern char vv_suffix[], vv_trailer[];
 #include <stdexcept>
 #include <unordered_map>
 #include <unordered_set>
+#include <array>
 #include <vector>
 #include <iomanip>
 #include <set>
```

yurivict, 2022-04-30:
> Does adding the the following line fix the problem?

array-related failures are gone, but there are still warnings/errors:
```
CCvcDb_init.cc:346:2: warning: 'register' storage class specifier is deprecated and incompatible with C++17 [-Wdeprecated-register]
        register netId_t mySourceNet, myDrainNet, myGateNet; //, myBulkNet;
        ^~~~~~~~~
CCvcDb_init.cc:346:2: warning: 'register' storage class specifier is deprecated and incompatible with C++17 [-Wdeprecated-register]
        register netId_t mySourceNet, myDrainNet, myGateNet; //, myBulkNet;
        ^~~~~~~~~
CCvcDb_init.cc:346:2: warning: 'register' storage class specifier is deprecated and incompatible with C++17 [-Wdeprecated-register]
        register netId_t mySourceNet, myDrainNet, myGateNet; //, myBulkNet;
        ^~~~~~~~~
CCvcDb_interactive.cc:1200:23: warning: equality comparison result unused [-Wunused-comparison]
                                        mySearchInstance == FindNetInstance(myNet, theCurrentInstanceId);
                                        ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
CCvcDb_interactive.cc:1200:23: note: use '=' to turn this equality comparison into an assignment
                                        mySearchInstance == FindNetInstance(myNet, theCurrentInstanceId);
                                                         ^~
                                                         =
CCvcDb_print.cc:555:49: warning: adding 'modelType_t' to a string does not append to the string [-Wstring-plus-int]
                        throw EDatabaseError("Invalid device type: " + theConnections.device_p->model_p->type);
                                             ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
CCvcDb_print.cc:555:49: note: use array indexing to silence this warning
                        throw EDatabaseError("Invalid device type: " + theConnections.device_p->model_p->type);
                                                                     ^
                                             &                       [                                       ]
CCvcDb_print.cc:585:49: warning: adding 'modelType_t' to a string does not append to the string [-Wstring-plus-int]
                        throw EDatabaseError("Invalid device type: " + theConnections.device_p->model_p->type);
                                             ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
CCvcDb_print.cc:585:49: note: use array indexing to silence this warning
                        throw EDatabaseError("Invalid device type: " + theConnections.device_p->model_p->type);
                                                                     ^
                                             &                       [                                       ]
CCvcDb_print.cc:707:61: warning: adding 'terminal_t' to a string does not append to the string [-Wstring-plus-int]
                default: { throw EDatabaseError("Invalid terminal type: " + theTerminal); }
                                                ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~
CCvcDb_print.cc:707:61: note: use array indexing to silence this warning
                default: { throw EDatabaseError("Invalid terminal type: " + theTerminal); }
                                                                          ^
                                                &                         [            ]
CCvcDb_print.cc:753:61: warning: adding 'terminal_t' to a string does not append to the string [-Wstring-plus-int]
                default: { throw EDatabaseError("Invalid terminal type: " + theTerminal); }
                                                ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~
CCvcDb_print.cc:753:61: note: use array indexing to silence this warning
                default: { throw EDatabaseError("Invalid terminal type: " + theTerminal); }
                                                                          ^
                                                &                         [            ]
mv -f .deps/CCvcDb_main.Tpo .deps/CCvcDb_main.Po
c++ -DHAVE_CONFIG_H -I. -I..   -fno-omit-frame-pointer -I/usr/local/include  -O2 -pipe -fno-omit-frame-pointer -fstack-protector-strong -fno-strict-aliasing -fno-omit-frame-pointer   -std=gnu++11 -MT CCvcDb_utility.o -MD -MP -MF .deps/CCvcDb_utility.Tpo -c -o CCvcDb_utility.o CCvcDb_utility.cc
mv -f .deps/CCvcDb_error.Tpo .deps/CCvcDb_error.Po
c++ -DHAVE_CONFIG_H -I. -I..   -fno-omit-frame-pointer -I/usr/local/include  -O2 -pipe -fno-omit-frame-pointer -fstack-protector-strong -fno-strict-aliasing -fno-omit-frame-pointer   -std=gnu++11 -MT CCvcParameters.o -MD -MP -MF .deps/CCvcParameters.Tpo -c -o CCvcParameters.o CCvcParameters.cc
mv -f .deps/CCvcDb.Tpo .deps/CCvcDb.Po
c++ -DHAVE_CONFIG_H -I. -I..   -fno-omit-frame-pointer -I/usr/local/include  -O2 -pipe -fno-omit-frame-pointer -fstack-protector-strong -fno-strict-aliasing -fno-omit-frame-pointer   -std=gnu++11 -MT CDevice.o -MD -MP -MF .deps/CDevice.Tpo -c -o CDevice.o CDevice.cc
4 warnings generated.
mv -f .deps/CCvcDb_print.Tpo .deps/CCvcDb_print.Po
c++ -DHAVE_CONFIG_H -I. -I..   -fno-omit-frame-pointer -I/usr/local/include  -O2 -pipe -fno-omit-frame-pointer -fstack-protector-strong -fno-strict-aliasing -fno-omit-frame-pointer   -std=gnu++11 -MT CEventQueue.o -MD -MP -MF .deps/CEventQueue.Tpo -c -o CEventQueue.o CEventQueue.cc
mv -f .deps/CDevice.Tpo .deps/CDevice.Po
[... cut ...]
2 errors generated.
gmake[5]: *** [Makefile:550: CVirtualNet.o] Error 1
gmake[5]: *** Waiting for unfinished jobs....
mv -f .deps/resource.Tpo .deps/resource.Po
mv -f .deps/mmap_file_pool.Tpo .deps/mmap_file_pool.Po
utility.cc:33:2: warning: 'register' storage class specifier is deprecated and incompatible with C++17 [-Wdeprecated-register]
        register void *value = malloc (size);
        ^~~~~~~~~
mv -f .deps/CPower.Tpo .deps/CPower.Po
mv -f .deps/CModel.Tpo .deps/CModel.Po
1 warning generated.
```

d-m-bailey, 2022-05-02:
@yurivict I've made changes to the source to handle most of the warning/errors. They're on the `clang-13` branch.
```
git clone -b clang-13 https://github.com/d-m-bailey/cvc.git

```
There are probably still problems with `src/mmappable_vector.h` because it has 4 compiler specific directives.
For example:
```
#ifdef __GNUC__
#if __GNUC__ == 3
                        A *the_allocator = static_cast<A*>(&(this->Base::_M_impl));
#else
                        A *the_allocator = &Base::_M_get_Tp_allocator();
#endif
#else
#error "Not GNU C++, please either implement me or use GCC"
#endif
```

Do you know how to implement this for the clang-13 compiler?

yurivict, 2022-05-03:
> Do you know how to implement this for the clang-13 compiler?

No.

yurivict, 2022-09-03:
What does this code achieve? How are regular containers insufficient?

yurivict, 2022-09-03:
STL vector generally can't be mmapable because it can generally resize any time.
If you need a deep modification like this - then you would be better off writing your own class from scratch.

yurivict, 2022-09-03:
The project fails to build with gcc-12 as well: https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=266189

d-m-bailey, 2022-09-03:
Thanks for the information, the gcc-12 fail looks like it has to do with the cdlParser.c which is bison output.

As for the mmap, I think I was trying to reduce the real memory required by essentially forcing the data to disk. I think the array sizes were fixed when allocated.

yurivict, 2022-09-04:
Yes, it's never a good idea to use STL container inner structures, because there are many implementations of STL. Having a custom class for mmapped vector seems like the way to go.

Currently the cvc's FreeBSD port doesn't build with clang, and it also prevents GCC upgrade to 12 because it fails with gcc-12.

lanfanb, 2023-06-14:
> @yurivict Does adding the the following line fix the problem?
>
> ```
> diff --git a/src/Cvc.hh b/src/Cvc.hh
> index 8df45c1..c01d04b 100755
> --- a/src/Cvc.hh
> +++ b/src/Cvc.hh
> @@ -59,6 +59,7 @@ extern char [...]

FYI
This maybe off-topic (not related to clang).
But I can confirm that: include <array> fixes the cdlParser.cc fail on gcc 13.1.0.
Maybe this fix also works for gcc-12, but sorry I don't have gcc-12.

yurivict, 2025-03-08:
The  latest release still fails in clang on access to Base::_M_impl._M_finish.

Could you please provide a reference where is Base::_M_impl._M_finish defined in the documentation?

d-m-bailey, 2025-03-08:
@yurivict Thanks for the feedback. With taxes and what not, it may be a week until I can look into it, but appreciate the reminder.