d-m-bailey/cvc #241: pyinstaller throwing error

RTimothyEdwards, 2022-04-25:
When running "make install" on CVC (in Fedora, using Python 3.10.2), I get the following error:

```
Making install in src_py
make[1]: Entering directory '/home/tim/gits/cvc/src_py'
pyinstaller -F check_cvc.spec --clean
24 INFO: PyInstaller: 5.0
24 INFO: Python: 3.10.2
25 INFO: Platform: Linux-5.16.9-200.fc35.x86_64-x86_64-with-glibc2.34
option(s) not allowed:
  --onedir/--onefile
makespec options not valid when a .spec file is given
make[1]: *** [Makefile:451: check_cvc] Error 1
```

Any ideas on what causes that, or what can be done about it?

---Tim

d-m-bailey, 2022-04-26:
I'm going to remove pyinstaller from the standard installation. It's only relevant for the GUI which is not critical for CVC error detection.

RTimothyEdwards, 2022-04-26:
Is there a Makefile recipe that doesn't build the GUI?

d-m-bailey, 2022-04-26:
There is currently not a recipe that does not try to build the GUI, `check_cvc`. However, you can ignore the error and proceed with `make install` to install cvc and the other scripts.

donnie-j, 2022-04-29:
This is the patch I use to bypass it, when building it for (and outside of) OpenLane.  There is a problem building with kivy, just make sure it's not installed and CVC will build.

```
patch -p1 << EOF
diff --git a/src_py/Makefile.am b/src_py/Makefile.am
--- a/src_py/Makefile.am
+++ b/src_py/Makefile.am
@@ -12,7 +12,7 @@
 all : check_cvc

 check_cvc : \$(checkcvc_sources) check_cvc.spec
-       pyinstaller -F check_cvc.spec --clean
+       pyinstaller check_cvc.spec --clean
        cp dist/check_cvc .

 install : check_cvc
EOF
```

RTimothyEdwards, 2022-04-29:
@donnie-j :  Excellent advice, worked like a charm!

RTimothyEdwards, 2022-04-29:
@d-m-bailey :  Is the `-F` option to pyinstaller necessary, or can it just be removed?

d-m-bailey, 2022-04-29:
The -F option creates a single executable file with all the python dependencies included. Removing the -F option copies all the necessary files to a build directory. This program (the GUI) is currently not configured to work with sky130, so even if the make run ends with a pyinstaller error, the necessary programs can be installed. I'm working on a patch that will remove the GUI installation by default. Should be in tomorrow.

d-m-bailey, 2022-04-30:
@RTimothyEdwards @donnie-j I removed the gui installation from the default. Please let me know if you have further problems.

closed 2022-04-30 by d-m-bailey