d-m-bailey/cvc #257: CVC build fails in OpenLane

hpretl, 2022-10-23:
When trying to update the CVC version used in OpenLane the CI build fails (Ubuntu 20.04.5) with this error message:

```
/bin/sh ../ylwrap cdlParser.yy y.tab.c cdlParser.cc y.tab.h `echo cdlParser.cc | sed -e s/cc$/hh/ -e s/cpp$/hpp/ -e s/cxx$/hxx/ -e s/c++$/h++/ -e s/c$/h/` y.output cdlParser.output -- bison -y -d -Wno-yacc 
[825](https://github.com/hpretl/OpenLane/actions/runs/3306748960/jobs/5457871414#step:3:855)
/cvc/src/cdlParser.yy:27.9-24: error: %define variable 'api.parser.class' is not used
[826](https://github.com/hpretl/OpenLane/actions/runs/3306748960/jobs/5457871414#step:3:856)
 %define api.parser.class {CCdlParser}
[827](https://github.com/hpretl/OpenLane/actions/runs/3306748960/jobs/5457871414#step:3:857)
         ^^^^^^^^^^^^^^^^
[828](https://github.com/hpretl/OpenLane/actions/runs/3306748960/jobs/5457871414#step:3:858)
make[1]: Leaving directory '/cvc/src'
[829](https://github.com/hpretl/OpenLane/actions/runs/3306748960/jobs/5457871414#step:3:859)
make[1]: *** [Makefile:546: cdlParser.cc] Error 1
[830](https://github.com/hpretl/OpenLane/actions/runs/3306748960/jobs/5457871414#step:3:860)
make: *** [Makefile:390: install-recursive] Error 1
```

d-m-bailey, 2022-10-24:
@hpretl Thanks for following up with this. I have a PR submitted to openlane, but it's not passing the CI and I have to investigate.

hpretl, 2022-10-24:
@donn has implemented a fix in https://github.com/The-OpenROAD-Project/OpenLane/pull/1440/files#diff-342331321bfc81580492bb90e88504b9f91888e1cbd60c5f78c6e2578d9615b0, I guess this solved the build issue:

`sed -i 's/api.parser.class/parser_class_name/' src/cdlParser.yy`

donn, 2022-10-24:
An issue I was only too familiar with, sadly. :) 
[image]

hpretl, 2022-11-17:
Fixed with proposed WA, let’s close it.

closed 2022-11-17 by hpretl