d-m-bailey/cvc #262: Build fail on Ubuntu 24.04 LTS

hpretl, 2024-12-27:
Using a build script that works with Ubuntu 22.04 LTS on 24.04 LTS, the following build fail happens:
```
make  all-recursive
make[1]: Entering directory '/tmp/cvc'
Making all in src
make[2]: Entering directory '/tmp/cvc/src'
make  all-am
make[3]: Entering directory '/tmp/cvc/src'
g++ -DHAVE_CONFIG_H -I. -I..     -O3 -std=gnu++11 -MT cdlParser.o -MD -MP -MF .deps/cdlParser.Tpo -c -o cdlParser.o cdlParser.cc
In file included from /usr/include/c++/13/vector:66,
                 from Cvc.hh:62,
                 from cdlParser.yy:32,
                 from cdlParser.cc:41:
/usr/include/c++/13/bits/stl_vector.h: In instantiation of ‘std::_Vector_base<_Tp, _Alloc>::~_Vector_base() [with _Tp = std::array<unsigned int, 5>; _Alloc = std::allocator<std::array<unsigned int, 5> >]’:
/usr/include/c++/13/bits/stl_vector.h:531:7:   required from here
/usr/include/c++/13/bits/stl_vector.h:370:49: error: invalid use of incomplete type ‘struct std::array<unsigned int, 5>’
  370 |                       _M_impl._M_end_of_storage - _M_impl._M_start);
      |                       ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~
In file included from /usr/include/c++/13/bits/hashtable_policy.h:34,
                 from /usr/include/c++/13/bits/hashtable.h:35,
                 from /usr/include/c++/13/bits/unordered_map.h:33,
                 from /usr/include/c++/13/unordered_map:41,
                 from CvcTypes.hh:33,
                 from Cvc.hh:39:
/usr/include/c++/13/tuple:2019:45: note: declaration of ‘struct std::array<unsigned int, 5>’
 2019 |   template<typename _Tp, size_t _Nm> struct array;
      |                                             ^~~~~
In file included from /usr/include/c++/13/bits/char_traits.h:50,
                 from /usr/include/c++/13/string:42,
                 from CvcTypes.hh:28:
/usr/include/c++/13/type_traits: In instantiation of ‘struct std::is_destructible<std::array<unsigned int, 5> >’:
/usr/include/c++/13/bits/stl_construct.h:188:51:   required from ‘void std::_Destroy(_ForwardIterator, _ForwardIterator) [with _ForwardIterator = array<unsigned int, 5>*]’
/usr/include/c++/13/bits/alloc_traits.h:948:20:   required from ‘void std::_Destroy(_ForwardIterator, _ForwardIterator, allocator<_T2>&) [with _ForwardIterator = array<unsigned int, 5>*; _Tp = array<unsigned int, 5>]’
/usr/include/c++/13/bits/stl_vector.h:735:15:   required from ‘std::vector<_Tp, _Alloc>::~vector() [with _Tp = std::array<unsigned int, 5>; _Alloc = std::allocator<std::array<unsigned int, 5> >]’
CCircuit.hh:35:7:   required from here
/usr/include/c++/13/type_traits:979:52: error: static assertion failed: template argument must be a complete class or an unbounded array
  979 |       static_assert(std::__is_complete_or_unbounded(__type_identity<_Tp>{}),
      |                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/c++/13/type_traits:979:52: note: ‘std::__is_complete_or_unbounded<__type_identity<array<unsigned int, 5> > >((std::__type_identity<std::array<unsigned int, 5> >(), std::__type_identity<std::array<unsigned int, 5> >()))’ evaluates to false
In file included from /usr/include/c++/13/bits/alloc_traits.h:33,
                 from /usr/include/c++/13/ext/alloc_traits.h:34,
                 from /usr/include/c++/13/bits/basic_string.h:39,
                 from /usr/include/c++/13/string:54:
/usr/include/c++/13/bits/stl_construct.h: In instantiation of ‘void std::_Destroy(_ForwardIterator, _ForwardIterator) [with _ForwardIterator = array<unsigned int, 5>*]’:
/usr/include/c++/13/bits/alloc_traits.h:948:20:   required from ‘void std::_Destroy(_ForwardIterator, _ForwardIterator, allocator<_T2>&) [with _ForwardIterator = array<unsigned int, 5>*; _Tp = array<unsigned int, 5>]’
/usr/include/c++/13/bits/stl_vector.h:735:15:   required from ‘std::vector<_Tp, _Alloc>::~vector() [with _Tp = std::array<unsigned int, 5>; _Alloc = std::allocator<std::array<unsigned int, 5> >]’
CCircuit.hh:35:7:   required from here
/usr/include/c++/13/bits/stl_construct.h:188:51: error: static assertion failed: value type is destructible
  188 |       static_assert(is_destructible<_Value_type>::value,
      |                                                   ^~~~~
/usr/include/c++/13/bits/stl_construct.h:188:51: note: ‘std::integral_constant<bool, false>::value’ evaluates to false
/usr/include/c++/13/bits/stl_construct.h:195:25: error: invalid use of incomplete type ‘std::iterator_traits<std::array<unsigned int, 5>*>::value_type’ {aka ‘struct std::array<unsigned int, 5>’} [-fpermissive]
  195 |       std::_Destroy_aux<__has_trivial_destructor(_Value_type)>::
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/c++/13/tuple:2019:45: note: declaration of ‘std::iterator_traits<std::array<unsigned int, 5>*>::value_type’ {aka ‘struct std::array<unsigned int, 5>’}
 2019 |   template<typename _Tp, size_t _Nm> struct array;
      |                                             ^~~~~
make[3]: *** [Makefile:550: cdlParser.o] Error 1
make[3]: Leaving directory '/tmp/cvc/src'
make[2]: *** [Makefile:392: all] Error 2
make[2]: Leaving directory '/tmp/cvc/src'
make[1]: *** [Makefile:403: all-recursive] Error 1
make[1]: Leaving directory '/tmp/cvc'
make: *** [Makefile:344: all] Error 2
```

hpretl, 2024-12-27:
@d-m-bailey Hi Mitch, I am working on upgrading IIC-OSIC-TOOLS to 24.04 and are currently stuck with the above fail. Could you please take a look?

hpretl, 2024-12-27:
It looks like it is related to this one https://github.com/d-m-bailey/cvc/issues/236

d-m-bailey, 2024-12-27:
@hpretl Thanks for the fix. I'll put this change in tomorrow.

d-m-bailey, 2024-12-28:
@hpretl I merged a fix, but I haven't tested or upped the version.